carrascoacd/ArduinoSIM800L #5: Does not work with Multi-Network SIMs

ChrisBingham1, 2017-03-17:
Hi,

Firstly, a great bit of serial feedback to COM6 when doing things in your code - something so often missed out and something which helped me understand what was going wrong!

Multi-Network SIMs - often used for MMI applications are normally Roaming SIMs. I have one of these in my Arduino SIM800L. This means that even though you are connected, the returned status is not :
+CREG:0,1 
it is:
+CREG:0,5

Therefore, in http.cpp since you have the following defined:
#define CONNECTED "+CREG: 0,1" 

This While statement causes it to be stuck in the loop:

Result HTTP::configureBearer(const char *apn){

  Result result = SUCCESS;

  unsigned int attempts = 0;
  unsigned int MAX_ATTEMPTS = 10;

  sendATTest();
// CONNECTED IN THE NEXT LINE FAILS THE ROAMING SIMS:
  while (sendCmdAndWaitForResp(REGISTRATION_STATUS, CONNECTED, 2000) != TRUE && attempts < MAX_ATTEMPTS){
    sendCmdAndWaitForResp(SIGNAL_QUALITY, CONNECTED, 1000);
    attempts ++;
    delay(1000 * attempts);
    if (attempts == MAX_ATTEMPTS) {
      attempts = 0;
      preInit();
    }
  }

So it looks like that the code needs to be tweaked so that if CONNECTED fails then it looks for ROAMING which would be defined as:
#define ROAMING "+CREG: 0,5" 

I've currently just cloned and changed CONNECTED in my copy to 0,5 and the code then works fine for my Roaming Multi Network SIM. 

Cheers,

Chris

carrascoacd, 2017-03-17:
Hi,

Ok, thank you for your contribution! I'll try to improve the "connected" concept by adding the Roaming Multi Network SIM state when I'd have a bit of free time.

Cheers

jaka87, 2019-05-16:
Had the same problem, took me about half an hour to figure out the solution. I came here to write about it so maybe I can save someone's time but its already documented.
I would just add that using roaming is not that uncommon. Most of telephone operators offer large data packages which are overkill and cost more than we are willing to spend to use few MB per month. Some charge insane amount for 1MB with prepaid options, others are reasonable but are using crazy billing unit. I stumble upon one operator with billing unit of 1MB!!!! If you want cheap data search for "IOT sim card".  Im my case i manage to cut down cost for 1MB for more than 50% comparing most optimal package while using the same network. If you are using IOT sim card (which I recommend to everybody) you are probably using roaming in at least 90% cases

carrascoacd, 2019-05-16:
@jaka87 I just created a PR with some work, I've tested a bit but I didn't release it https://github.com/carrascoacd/ArduinoSIM800L/pull/28

jaka87, 2019-05-18:
Great work. I tested it for more than one day with update intervals of about 5min. Everything seems fine.