carrascoacd/ArduinoSIM800L #7: Isolate HTTP JSON data

jLynx, 2017-07-21:
Is there a way to Isolate the JSON data? Right now I am getting
`+HTTPACTION: 1,200,{"hello":true}`
where all I am wanting is 
`{"hello":true}`

Also how do I turn off all the console output from the module?

Thanks

carrascoacd, 2017-07-21:
Hi!,

In order to turn off all the console output you have a flag in the HTTP class constructor, the last one:
`const bool DEBUG = false;
const HTTP http(9600, RX_PIN, TX_PIN, RST_PIN, DEBUG);`

`+HTTPACTION: 1,200,{"hello":true}` is the console log you are getting, but the response the library returns is already isolated, for example:
`http.post("your.api", body, response);`
with the debug flag=true (that by default is true in the HTTP constructor) will log 
`+HTTPACTION: 1,200,{"hello":true}` but the `response` variable will contain `{"hello":true}`
Check [this example](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/HttpExample/WeatherStationExample.ino)  where I'm using ArduinoJson to parse JSON data into an object also.

jLynx, 2017-07-21:
Awesome! Thanks for the fast reply!

closed 2017-07-21 by jLynx