carrascoacd/ArduinoSIM800L #8: Hardware Serial

Travis92, 2018-02-25:
Hi, i am currently working with your library and i would like to implement it with HW serial2 on a Mega.
Are you working for developing it? 

Thanks

P.

carrascoacd, 2018-02-25:
Hi,

No, I'm not working at the moment in this feature. You are very welcome to
contribute to the library

Thanks!

ankgt, 2018-03-10:
@Travis92 I am not sure about the Mega, but it was very easy to do on my ESP32 based board. Just modified a couple of lines in Sim800.h to use HardwareSerial instead of SoftwareSerial. 

Just make sure to comment out the serialSim800.listen() line in Sim800.c sendCmd() as it is not needed for HardwareSerial.

fedy0, 2018-05-19:
@ankgt

Could open a pull request or could you share your esp32 based code for this same library?

Thanks in advance

ankgt, 2018-05-22:
@fedy0 

I have made a lot of changes to the library so it will be confusing if I post it. I believe the main change that is required to work with the ESP32 is very simple and it is in Sim800.h:

`
SIM800(unsigned int baudRate, unsigned int rxPin, unsigned int txPin, unsigned int rstPin, bool debug):serialSIM800(1) {
        serialSIM800.begin(baudRate, SERIAL_8N1, rxPin, txPin, false); 
        debugMode = debug;
        resetPin = rstPin;
    };
`

Also change the following at the end of Sim800.h:
Remove: `SoftwareSerial serialSIM800`
Add: `HardwareSerial serialSIM800`

You will have to change the definitions for Http.h also to accept the rxPin and txPin.  I don't believe anything else was required.

There are some small bugs in the library which cause it to crash in some cases, and I do plan to open a separate thread for that, just not got a change to get to it. Will do it shortly.

Hope it helps.

kierenblack, 2019-05-19:
@ankgt How did you get on with this? Did you manage to post your code somewhere?

ankgt, 2019-05-20:
@kierenblack  I have changed my modem to Quectel M66 instead of the Simcom, but continued using this base library. However, I have heavily customised it my specific requirements.

Where are you getting stuck?