carrascoacd/ArduinoSIM800L #9: not working

ran-j, 2018-03-02:
Im traing to send data to my server

`#include <Http.h>
int tes = 0;
int RX_PIN = 7;
int TX_PIN = 8;
int RST_PIN = 12;
HTTP http(9600, RX_PIN, TX_PIN, RST_PIN);
void setup() {
  Serial.begin(9600);
  Serial.println("Starting!");
}

void loop() {
  if(tes==0){
    tes = 1;
    http.configureBearer("movistar.es");
    http.connect();

    char response[256];
    Result result = http.post("www.rannet.com.br/test", "{\"date\":\"12345678\"}", response);

    Serial.println(response);

    http.disconnect();
  }else{

    
  }

}`

And the output is 

`Starting!
AT

OK
AT+CREG?

+CREG: 0,4

OK
AT+CBC

+CBC: 0,0,3242

OK
AT+CSQ

+CSQ: 22,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3247

OK
AT+CSQ

+CSQ: 19,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3264

OK
AT+CSQ

+CSQ: 19,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3258

OK
AT+CSQ

+CSQ: 17,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3253

OK
AT+CSQ

+CSQ: 18,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3231

OK
AT+CSQ

+CSQ: 22,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3264

OK
AT+CSQ

+CSQ: 22,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3264

OK
AT+CSQ

+CSQ: 23,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CBC

+CBC: 0,0,3264

OK
AT+CSQ

+CSQ: 23,0

OK
`

carrascoacd, 2018-03-03:
Hi!

It looks like you are not connected to any network as AT+CREG returns 0,0.
Check this out http://m2msupport.net/m2msupport/atcreg-network-registration/

It could be a problem related to your network provider. Check if the SIM
card is working in a mobile phone first.

Note: you must disable the PIN code in your SIM card as the library does
not suport PIN verification for the moment.

Cheers,

ran-j, 2018-03-05:
how do I do AT+CRE registration ?

ran-j, 2018-03-05:
Right, now this : 

`
AT+CREG?

+CREG: 0,2

OK
 ⸮AT+CSQ

+CSQ: 0,0

OK
AT+CREG?

+CREG: 0,2

OK

Call Ready

SMS Ready
AT+CBC

+CBC: 0,0,3286

OK
AT+CSQ

+CSQ: 0,0

OK
 ⸮AT+CBC

+CBC: 0,0,3209

OK
AT+CSQ

+CSQ: 0,0

OK

Call Ready

SMS Ready
AT+CREG?

+CREG: 0,2

OK
`

carrascoacd, 2018-03-06:
I think your device is restarting so you are reconnecting always, notice the � characters and the "Call Ready" output. Check the power supply as it could not be enough (should deliver at least 2A peak current).

ran-j, 2018-03-13:
I made it, apparently dont work with DCcduino.

carrascoacd, 2018-03-13:
@ran-j I'm using a DCcduino uno also and it works perfectly. I had a similar problem and it was because the power supply for the SIM800L was not enough (it can'n be connected directly to the arduino, it needs an independent LiPo battery)

closed 2018-03-13 by carrascoacd