carrascoacd/ArduinoSIM800L #10: Getting Post Error

rizban114406, 2018-03-08:
Every Time I am Sending data to the server I am getting this response +HTTPACTION: 1,400,169. Can you please tell me the exact syntax of sending a single value. my api receives a single value.

carrascoacd, 2018-03-08:
Hi,

You are receving a Bad Request error (400) from your server, it means the
request is malformed syntax.
Show me the code that calls the post() method in arduino

rizban114406, 2018-03-12:
Hello,
Hope You are doing well. I will get back to you about post method after
some time. currently I am using GET method to send data. Have not used your
library as I have a very short time to complete my project. So For initial
release I am not using your library. As you are expert in this Please help
me out on a problem related to Get Method.

AT+HTTPPARA="URL","
http://myServerIpAddess/carcopolo_arduino/Json_index.php?tag=insertdevicedata&data={
"data":[{"userid":1,"trip_reference":12434243,"latitude":"23.7907832","longitude":"90.4090195","date":"20160510","time":"12:10:02","speed":"100","distance":"100"}]}"

This is the command i am passing. But every time I am getting ERROR
response. I have generated the json string using ArduinoJson Library. and
Then stored this json string to a character array like this. (
root.printTo(body);)

I have copied this full command from the Serial monitor. Still I am getting
error. If I directly post this url part to the browser it is returning a
proper response. Can you tell me what is the problem here? Your help is
really appreciated. Thanks in advance. my api receives 2 parameters,
tag & data.

Best Regards,
Sayed Rizban Hussain

rizban114406, 2018-03-12:
One Finding,
http://myServerIpAddess/carcopolo_arduino/Json_index.php?tag=insertdevicedata&data={
"data":[{\"userid\":1,\"trip_reference\":12434243,\"latitude\":\"23.7907832\",\"longitude\":\"90.4090195\",\"date\":\"20160510\",\"time\":\"12:10:02\",\"speed\":\"100\",\"distance\":\"100\"}]}"

if i send it like this it is returning a OK response. how can I make my json array like this?? I have just used root.printTo(body) function to copy jsonobject to my character array.

rizban114406, 2018-03-12:
[image]

carrascoacd, 2018-03-13:
Hi,

To do a get request you need to url encode your params https://en.wikipedia.org/wiki/Query_string. 
But I suggest to use a post request for this case as is the way of doing it in other case you will have a lot of problems.
I'm going to close this issue as is not related to the library, if you need help, write me an email.

closed 2018-03-13 by carrascoacd

rizban114406, 2018-03-14:
Dear,

Thank you so much for your response. I have downloaded you library. And
prepared a Sample Json data using ArduinoJson Library. Then for sending it
to the server I have copied it to a Character Array (body). My API receives
two parameters (tag, data). For your better understanding I am attaching a
screen shot of Postman where I have tested my api. The code that I have
created is Also given below,

Code:

#include <ArduinoJson.h>
#include <Http.h>

void setup() {
  Serial.begin(9600);
  delay(4000);
  char body[200];

  DynamicJsonBuffer jBuffer;
  JsonObject& root  = jBuffer.createObject();
  JsonObject& root2  = jBuffer.createObject();
  JsonArray& data  = root.createNestedArray("data");
  root2["userid"] = 1;
  root2["trip_reference"] = 12434243;
  root2["latitude"] = "23.7907832";
  root2["longitude"] = "90.4090195";
  root2["date"] = "20160510";
  root2["time"] = "12:10:02";
  root2["speed"] = "100";
  root2["distance"] = "100";
  data.add(root2);

  root.printTo(body,sizeof(body));

  unsigned int RX_PIN = 7;
  unsigned int TX_PIN = 8;
  unsigned int RST_PIN = 12;
  HTTP http(9600, RX_PIN, TX_PIN, RST_PIN);
  http.configureBearer("movistar.es");
  http.connect();

  char response[256];
  Result result =
http.post("IP_Address/carcopolo_arduino/Json_index.php",body, response);

  Serial.println(response);

  http.disconnect();

}

void loop() {
  // put your main code here, to run repeatedly:

}

Sample Data :

tag =   insertdevicedata
data = {"data": [{"userid": 1,"trip_reference": 12434243,"latitude":
"23.7907832","longitude": "90.4090195","date": "20160510","time":
"12:10:02","speed": "100","distance": "100"}}]}

Can You please guide me what changes should i made to send this data using
your library? You have called your post method like this,

Result result =
http.post("IP_Address/carcopolo_arduino/Json_index.php",body, response);

what will be the exact structure/syntax of my body variable?

Really looking forward to your reply. Please guide me with an example if
possible. That would be a great help for me.

Thank you,
Sayed Rizban Hussain

carrascoacd, 2018-03-14:
An example: https://github.com/carrascoacd/ArduinoSIM800L/blob/master/HttpExample/WeatherStationExample.ino