carrascoacd/ArduinoSIM800L #19: AT+HTTPINIT   -> ERROR

farsajik, 2019-02-05:
```
#include <Http.h>

#define RX_PIN 5
#define TX_PIN 6
#define RST_PIN 10
#define DEBUG TRUE
unsigned long lastActionTime = 0;
const HTTP http(9600, RX_PIN, TX_PIN, RST_PIN, DEBUG);

void setup() {
  // put your setup code here, to run once:
    if (DEBUG){
    Serial.begin(9600);
    while(!Serial);
    Serial.println(F("Starting!"));
  }

}

void loop() {
  // put your main code here, to run repeatedly:
  if ( millis() > lastActionTime + 5000 ) {
    Serial.println("request");
      sendData();
  }
}

void sendData(){
  
  char response[32];
  char body[90];
  Result result;
  http.configureBearer("internet.bakcell.com");
  result = http.connect();
  result = http.post("http://www.iforce2d.net/test.php", "hello", response);
  //http.disconnect();
   Serial.println(response);
 
}

Starting!
request
AT

OK
AT+CREG?

+CREG: 0,1

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","internet.bakcell.com"

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"10.169.99.51"

OK
AT+HTTPINIT

ERROR
AT+HTTPPARA="CID",1

OK
AT+HTTPPARA="URL","http://www.iforce2d.net/test.php"

OK
AT+HTTPSSL=0

OK
AT+HTTPPARA="CONTENT","text/plain"

OK
AT+HTTPDATA=5,10000

DOWNLOAD

OK
AT+HTTPACTION=1
OK

+HTTPACTION: 1,200,34

```

ostaquet, 2019-02-06:
Hi @farsajik,
Did the error appear in each loop or only in the first one?
It is strange that the HTTPINIT generated an ERROR but the rest of the communication seems to work according to your logs.
According to the specs, the error should be an error related to mobile equipment functionality.
The output of HTTPINIT in ERROR should be something like `+CME ERROR: <err>`. This log will help to diagnose the issue.
KR,
Olivier

farsajik, 2019-02-06:
Thanks for you reply.error apears in each loop.another sketches from internet works fine with this module.but this library generates error.i dont know how to solve this

closed 2019-02-07 by farsajik