carrascoacd/ArduinoSIM800L #20: Data type??

farsajik, 2019-02-07:
Helli. can i send and receive STRINGS not Json with this library?

ostaquet, 2019-02-07:
Hi,
By using the post() method, you're able to send any strings and receive any strings.

`Result HTTP::post(const char *uri, const char *body, char *response)`

The return value Result is the status of the request (success, error...).
The content sent by HTTP is the body (jSON or text or XML or whatever).
The result from the server is stored in the response (whatever the data sent by the server).

You can also change the content type to send to the server by adjusting the definition:
`#define HTTP_CONTENT "AT+HTTPPARA=\"CONTENT\",\"application/json\"\r\n"`

KR,
Olivier

ostaquet, 2019-02-07:
Sorry, missed a point (it's not my code :-P)

The HTTP_CONTENT defines only the way to send the content to HTTP server.

If you need keep the response string as received, you should replace the jSON parsing method `parseJSONResponse()` by a simple memcpy inside the readResponse() method.

`void HTTP::readResponse(char *response){

  char buffer[128];
  cleanBuffer(buffer, sizeof(buffer));
  cleanBuffer(response, sizeof(response));

  if (readBuffer(buffer, sizeof(buffer)) == TRUE){
    **parseJSONResponse(buffer, sizeof(buffer), response);**
  }
}`

KR,
Olivier

farsajik, 2019-02-07:
thank you for reply.like this???

void HTTP::readResponse(char *response){

  char buffer[128];
  cleanBuffer(buffer, sizeof(buffer));
  cleanBuffer(response, sizeof(response));

  if (readBuffer(buffer, sizeof(buffer)) == TRUE){
    memcpy (buffer, response, sizeof(buffer));
  }
}

ostaquet, 2019-02-07:
Yep, it should work ;-)

carrascoacd, 2019-02-07:
@farsajik could be great to parse the response data based on the response headers instead of hardcode the method. 

To do it we should send the `AT+HTTPHEAD` command then we can parse the `Content-Type` header to decide what kind of parsing do, something like

```
sendCmdAndWaitForResp(¨AT+HTTPHEAD\r\n¨ OK, 2000)

if (strncmp(contentType, ¨application-json¨) == 0){
  // parse json
}
else {
 // do whatever
}
```

closed 2019-02-08 by farsajik