carrascoacd/ArduinoSIM800L #29: Crashes ESP32 for large size of json.

teezzan, 2019-06-14:
Hello, 
Thanks for an awesome library. 
I made a few modifications to the code to make it work on my ESP32. I tried to send a relatively large json. It crashes the microcontroller. I was able to send smaller ones with no problems at all. I'd like to ask if there is a limit to the size of json one can send and how to increase such value. 
Thanks once again for the awesome work.

Regards
Taiwo

carrascoacd, 2019-06-14:
Hi, it depends on your ESP32 I guess you have about  512KiB of available memory, which not all will be free for handling large amounts of data, in this case your JSON. 

Probably you are getting an out of memory and it crashes, 1  character is 1 byte so if you send about 512 characters you will be out of memory.
The solution is split the JSON into smaller ones, or try to crop it if you can. This library does not support streaming of response reads so you won't be able to handle large JSON.

teezzan, 2019-06-14:
My server receives JSON. Won't splitting it cause a need for server side corrections?

teezzan, 2019-06-29:
It turned out sim800l can't take that much stream of data. I finally had to split the json. Thanks for your help.

closed 2019-06-29 by teezzan