carrascoacd/ArduinoSIM800L #31: GEO location function explanation

jaka87, 2019-09-28:
Hi.
I want to share what i found out about GEO function calling AT+CIPGSMLOC command. At the time i committed the function I thought it triangulates location between different antennas but this is not the case. It just gives location of the closest network antenna. While i still think it is useful for many cases it wont work well as an anti theft device like I thought. 

One more thing. It is not advised to use it one every measurements. My weather station send data every 5 minutes and i included this function as well. What happens from time to time is that this function causes reset of arduino. It took me quite long time to figure it out. At first i blamed signal quality then temperature sensors, then i finally plugged my raspberry to get the serial data from the arduino. Im not sure why exactly this happens but most of the time its is related with bad weather including heavy rain. I must mention that i use 1000 uF cap and two smaller ceramic ones so power should not be issue but arduino gets reset. Even when I upgraded to 3300uF cap the problem still persisted. Sometimes arduino would reset every few days but in case of a bad weather resets would be even in interval of few minutes. After few resets the WS would become unresponsive. Upon inspection power LED was blinking and resenting arduino didn't help. So arduino program was bricked but I could plugged it to PC reupload program and it worked again. Then i burn program to arduino without the boatloader hopping this would fix the trouble. While the problem persisted, arduino however didn't need program reupload it was enough to cut the power and reset the board. I have been running my station without this function for last few days and i arduino didn't reset once. It is still early to say that issue is 100% resolved thou.

Im sharing this to help anybody struggling with same issue. It was difficult to pinpoint this issue since it can happen few times a day to a few weeks interval.

jaka87, 2019-09-29:
It turns out i might spoke prematurely. I swich 1.8k resistor for 4.7k on RX by mistake. This issue might have been caused by low voltage problem. Voltage on RX shold be 2.5-2.8V. In my case using the wrong resistor it is about 2.3V i guess. Will try different resistor and give update in next weeks...

carrascoacd, 2019-09-29:
Check also if you have available memory or maybe a memory leak

jaka87, 2019-10-02:
This is the first thing I checked. It is fine. Removing the GPS location function did make faster and more stable updates with no resets, thou the arduino did still hang after just 3 days of operation. It always happens on bad weather including heavy rain. I first thought it is becouse of bad signal, then faulty temperature sensor  but both were fine to. Interesting i have two stations one located 50m from my house the other 200km away. The one at home keeps working without problem the other one keeps hanging. The fastest malfunction was after just 1 day, the but it worked 1 month in summer. I tought maybe there is an error on PCB and took the one from home to away location. Same thing. Im kind of  putting my hopes in resistor change, if that wont work i will try to implement the watchdog and reset the station when necessary. I removed all the sensors except anemometer and will do further testing.

jaka87, 2019-10-03:
It might be related to arduino pro mini. There is a bootloader bug that puts arduino in endless loop with blinking power LED. More info here https://andreasrohner.at/posts/Electronics/How-to-make-the-Watchdog-Timer-work-on-an-Arduino-Pro-Mini-by-replacing-the-bootloader/ 
I tried the test sketch and it put arduino to same state that we find my weather station in. Same sample works fine on arduino mega. This might be it.

jaka87, 2019-11-24:
The bootloader bug definitely was a problem, I'm not relay sure if GEO location is even causing any problems or not but i still turned it off just in case, for now. As it turns out when I burned the sketch with programmer (to bypass bootloader) the default value for extended fuse is to disable BOD which is a big no no in my case. When voltage drooped, frequency decrease and time on arduino was running slower that it should, and therefore wind speed was measured much higher (over 100 knots) which is very unlikely. Station also stopped reporting data. Im not sure if it got stuck because voltage drop was to low and caused some glitch in program or because of faulty battery. I changed fuse setting to enable BOD and will also replace battery with a new one and then we will see how station will behave in low temperatures...

closed 2019-11-24 by jaka87

carrascoacd, 2019-11-24:
Thanks for sharing, always learning.

What is the battery voltage of the new battery and the old one?
I'm using a 18350 3,7V battery and it "works". It is true that the threshold of the sensors change and you have to take it into account since the reference voltage is 3,7 instead of 5V in my case.

I know there are 3.3V Arduino boards, maybe it helps you.

jaka87, 2019-12-07:
Both batteries were 3.7V and I use arduino 3.3V. The problem is probably due to the energy bursts of SIM800L. It might be faulty battery. We will change it (hopefully today) and will see then.

Since the beginning i hadn't have a problem with station on my balcony, but always the one 200km away so debugging is a bit tricky but we came far this year and I believe we are going to get stable device sooner or later...