carrascoacd/ArduinoSIM800L #33: dtmf

atulgo, 2019-12-30:
Detection of DTMF tones in my SIM800L module.

I get "OK" when I send 'AT+DDET=1,1000,1,1* ' over uart .
But it doesnt seem to detect anything afterwards.
Perhaps I am doing something wrong.
Would be great to get some feedback.

regards
Atul
ref : *SIM800 Series_AT Command Manual_V1.09

carrascoacd, 2019-12-30:
I can't help you with such information. Could you share your output and the code please?

atulgo, 2019-12-31:
Thanks for replying
I was just playing with AT commands using 'minicom' serial program :)
Further investigation led the conclusion that firmware doesnt support DTMF. 
Any suggestions in that area ?
regards

carrascoacd, 2019-12-31:
Ok!
I can't help you, I've never used this before

closed 2019-12-31 by atulgo