carrascoacd/ArduinoSIM800L #35: Cant build the v0.2

jaka87, 2020-03-01:
Hi there. 
I was playing around with my project and decided to update to latest version of your script.
i got some errors and i tough that maybe it's my code, but got the same with the test sample.

First i get 
```
home/jaka87/Arduino/libraries/ArduinoSIM800L/src/GPRS.cpp: In function '_GLOBAL__I_65535_0_sleep3.ino.cpp.o':
/home/jaka87/Arduino/libraries/ArduinoSIM800L/src/GPRS.cpp:45:12: error: variable 'OK_' with dynamic initialization put into program memory area
   45 | const char OK_ PROGMEM = "OK";
      |            ^
lto-wrapper: fatal error: /usr/bin/avr-gcc returned 1 exit status
compilation terminated.
/usr/bin/avr-ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
```

but then I tried to uncomment the GPRS and i then get 

```
/home/jaka87/Arduino/libraries/ArduinoSIM800L/src/Http.cpp: In member function 'Result HTTP::connect(const char*)':
/home/jaka87/Arduino/libraries/ArduinoSIM800L/src/Http.cpp:56:19: error: 'openGPRSContext' was not declared in this scope
   56 |   Result result = openGPRSContext(this, apn);
      |                   ^~~~~~~~~~~~~~~
/home/jaka87/Arduino/libraries/ArduinoSIM800L/src/Http.cpp: In member function 'Result HTTP::disconnect()':
/home/jaka87/Arduino/libraries/ArduinoSIM800L/src/Http.cpp:66:19: error: 'closeGPRSContext' was not declared in this scope
   66 |   Result result = closeGPRSContext(this);
      |                   ^~~~~~~~~~~~~~~~
Using library ArduinoJson at version 6.13.0 in folder: /home/jaka87/Arduino/libraries/ArduinoJson 
Using library ArduinoSIM800L at version 0.2 in folder: /home/jaka87/Arduino/libraries/ArduinoSIM800L 
Using library SoftwareSerial at version 1.0 in folder: /usr/share/arduino/hardware/archlinux-arduino/avr/libraries/SoftwareSerial 
exit status 1
Error compiling for board Arduino Pro or Pro Mini.

```

carrascoacd, 2020-03-01:
Interesting. What board are you using?

Thanks for your feeback!

jaka87, 2020-03-01:
Im using arduino pro mini, but I also tried compiling for nano and uno. 
I also tried using different Arduino IDE 1.8.12 and 1.8.9

All result in same error

carrascoacd, 2020-04-04:
@jaka87, could you try the branch of the PR https://github.com/carrascoacd/ArduinoSIM800L/pull/41? I fixed the errors. If it works, I'll release a new bugfix version.

closed 2020-04-18 by carrascoacd