carrascoacd/ArduinoSIM800L #39: Undefined conversions, unused variables & comparison between expressions

nabeelkirmani, 2020-03-24:
### Undefined conversion:
[line 45](https://github.com/carrascoacd/ArduinoSIM800L/blob/d0fcfffa7d8df7a1e81fbef712f5ca26fd2418bc/src/GPRS.cpp#L45) should be `const char OK_[] = "OK";`

### Unused variable
[char resp[16];](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Http.cpp#L112) in Http.cpp

### Comparisons between signed and unsigned expressions
[int i](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Parser.cpp#L36) in `parseATResponse()` could be `unsigned int i`
[int i = 0;](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Parser.cpp#L46) in `parseJSONResponse()` could be `unsigned int i = 0;`

carrascoacd, 2020-03-24:
Thanks for letting me know.

I'll pass a linter to the whole project also.

carrascoacd, 2020-04-04:
Suggestions coming soon... https://github.com/carrascoacd/ArduinoSIM800L/pull/41

closed 2020-04-04 by carrascoacd