carrascoacd/ArduinoSIM800L #40: How to post stream instead of upload to file via FTP.

rtarta, 2020-04-03:
Hello thanks for your works, is it possible in your code to post file as stream as below could you check it is possbile ?

https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.cpp#L583

carrascoacd, 2020-04-04:
Hi! thanks.

I think so you only have to fetch the bytes in chunks from the stream and use the example provided in the documentation.

```c
unsigned int RX_PIN = 7;
unsigned int TX_PIN = 8;
unsigned int RST_PIN = 12;
const char BEARER[] PROGMEM = "gprs-service.com";
const char FTP_SERVER[] PROGMEM = "ftp.server";
const char FTP_USER[] PROGMEM = "user";
const char FTP_PASS[] PROGMEM = "pass";

FTP ftp(9600, RX_PIN, TX_PIN, RST_PIN);
ftp.putBegin(BEARER, "example.txt", FTP_SERVER, FTP_USER, FTP_PASS);
ftp.putWrite("hello!", sizeof("hello!"));
ftp.putEnd();
```

Look at the `ftp.putWrite` function, it accepts as first argument bytes, and as the second argument the number of bytes. So you could use something like (I didn't test it):
```c
unsigned int RX_PIN = 7;
unsigned int TX_PIN = 8;
unsigned int RST_PIN = 12;
const char BEARER[] PROGMEM = "gprs-service.com";
const char FTP_SERVER[] PROGMEM = "ftp.server";
const char FTP_USER[] PROGMEM = "user";
const char FTP_PASS[] PROGMEM = "pass";

FTP ftp(9600, RX_PIN, TX_PIN, RST_PIN);

int readBytes = 100; // For this example
char buff[readBytes];

ftp.putBegin(BEARER, "example.txt", FTP_SERVER, FTP_USER, FTP_PASS);

while (stream.available()) {
  int bytesRead = stream.readBytes(buff, readBytes);
  ftp.putWrite("hello!", sizeof("hello!"));
}

ftp.putEnd();
```

Also, check [this](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/examples/WeatherStationExample.ino#L83) example where I do the same, but using a `File` object instead.

rtarta, 2020-04-13:
@carrascoacd Is it working with esp32 SIM800L
i could not inject your library :(

carrascoacd, 2020-04-14:
Could you share the logs of the error you are getting, please?

I think you are having problems because you are using the standard SoftwareSerial library which is not working on ESP32. Try using this https://github.com/plerup/espsoftwareserial instead, and let me know if it works for you :)

rtarta, 2020-04-14:
@carrascoacd Thanks for your helps, below i am getting error when use only this includings.  when i build this with out any serial at start i gets error.

`#include <Ftp.h>
#include <Geo.h>
#include <GPRS.h>
#include <Http.h>
#include <Parser.h>
#include <Result.h>
#include <Sim800.h>

void setup()
{

}

void loop() {

} `

Arduino: 1.8.12 (Windows 10), Board: "AI Thinker ESP32-CAM"

C:\Arduino\libraries\ArduinoSIM800L-master\src\Http.cpp:50:21: error: array must be initialized with a brace-enclosed initializer

 const char* OK_[] = "OK";

                     ^

C:\Arduino\libraries\ArduinoSIM800L-master\src\Http.cpp: In member function 'Result HTTP::setHTTPSession(const char*)':

C:\Arduino\libraries\ArduinoSIM800L-master\src\Http.cpp:186:46: error: no matching function for call to 'HTTP::sendCmdAndWaitForResp(char [128], const char* [3], int)'

   if (sendCmdAndWaitForResp(buffer, OK_, 2000) == FALSE)

                                              ^

In file included from C:\Arduino\libraries\ArduinoSIM800L-master\src\Http.h:31:0,

                 from C:\Arduino\libraries\ArduinoSIM800L-master\src\Http.cpp:28:

C:\Arduino\libraries\ArduinoSIM800L-master\src\Sim800.h:120:9: note: candidate: int SIM800::sendCmdAndWaitForResp(const char*, const char*, unsigned int)

     int sendCmdAndWaitForResp(const char *cmd, const char *resp, unsigned timeout);

         ^

C:\Arduino\libraries\ArduinoSIM800L-master\src\Sim800.h:120:9: note:   no known conversion for argument 2 from 'const char* [3]' to 'const char*'

C:\Arduino\libraries\ArduinoSIM800L-master\src\Ftp.cpp:48:21: error: array must be initialized with a brace-enclosed initializer

 const char* OK_[] = "OK";

                     ^

exit status 1
Error compiling for board AI Thinker ESP32-CAM.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

carrascoacd, 2020-04-14:
Thanks for sharing, the board you use is nice. How do you power it / what kind of battery do you use?

@rtarta please, please try this branch and check if it works. I don't have the real physical device, but it compiles TM https://github.com/carrascoacd/ArduinoSIM800L/pull/42

rtarta, 2020-04-16:
@carrascoacd  My friend, i made a little bir change on SIM800.h over SoftwareSerial. I turned it to HarwareSerial over any pin to connect Esp32Cam. But i can not manage to send image file over GPRS. 
Esp32 cam taking image and write it into SD card, when i start to read it i locks sd card reading. Now i do not knwo what to do.  if need i can share SIM800.h file. I have manage to send file over HttpPost via Wifi with esp32 you can check it from [there ](https://rahmitugrulaltin.blogspot.com/2020/04/esp32-camera-taking-picture-and-send-to.html). In this HttpLibrary i can set directly file in to post request.
I do not know what to do :(

carrascoacd, 2020-04-16:
Are you powering the SIM800L with a dedicated LiPo battery?

Show me your design please in case I can help you.

rtarta, 2020-04-16:
@carrascoacd  We are powering the sim800 with LM2596 . My circut is same as [this](url
[image]). This is arduino by my board is **ESP32Cam**.

I am using power supplier with LM2596.

carrascoacd, 2020-04-18:
Try this:

```c++
FTP ftp(9600, RX_PIN, TX_PIN, RST_PIN);
ftp.putBegin(BEARER, "example.txt", FTP_SERVER, FTP_USER, FTP_PASS);
ftp.putWrite("hello!", sizeof("hello!"));
ftp.putEnd();
```

And check if it creates a new file. Please, send me the logs of the serial monitor.

SIM800L should be powered with a LiPo battery of 3.7V because of the high current spikes, I think you are not powering it right and because of it, the execution blocks.

rtarta, 2020-04-18:
@carrascoacd  Dear Friend i manage to send file to remote server with other library(ArduinoHttpClient) but not ftp. With your library i could also create small data on remote ftp but my file size min 10 Kb. Thanks your supports.

carrascoacd, 2020-04-18:
Could you share the code you use to upload the file with my library?

pcbtborges, 2020-05-12:
Hi, thanks for your code.
I have a project where I have to transfer an image.jpg that is already loaded to a FAT partition in my ESP32. I have tried AT commands but it does not work all the time. Some 50% of the times less than 100% of the image arrives intact. 
Can you tell me if all I need is the function: 
`void uploadFile(const char *filename) {`
And library to do the transfer?
I am talking about the function from WeatherStationExample.ino
The image file is around 12KBytes big
Thanks in advance
pcborges@yahoo.com

carrascoacd, 2020-05-13:
Hi! you can follow this example where I read an image from the SD card and upload it in batches to the FTP server https://github.com/carrascoacd/ArduinoSIM800L/blob/master/examples/WeatherStationExample.ino#L83

You could increase the `unsigned int chunkSize = 64;` to a higher value so it transfers data faster, for example, set it to 512.

Important note. You will need enough SRAM as the FTP library will need at least 512 bytes (since it uses pages of that size to r/w to the SD card). I had many many problems with the 2KB that ATMega328p provides, it was not enough.

In order to make it works:
- If you don´t have enough memory random things could happen, ensure you have enough memory.
- Ensure your SIM800L is properly powered.

If you use ArduCAM mini, it has an internal buffer so you don´t need the SD card as a middleman.

pcbtborges, 2020-05-13:
Hi, thanks a lot for taking the time to respond to my question.

I have already created a sketch but it is not working. It freezes on ftp.putBegin( (see below)

My Sketch:
`#define DEBUG_ESP              //comment out to deactivate debug console

#ifdef DEBUG_ESP
  #define pDBGln(x) Serial.println(x)
  #define pDBG(x)   Serial.print(x)
#else 
  #define pDBG(...)
  #define pDBGln(...)
#endif

#include <Ftp.h>
#include "FFat.h"

#define GSM_BAUD_RATE 9600

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26

unsigned int RX_PIN = 7;
unsigned int TX_PIN = 8;
unsigned int RST_PIN = 12;
const char BEARER[]     PROGMEM = "claro.claro.com";
const char FTP_SERVER[] PROGMEM = "ftp.myftp.com";
const char FTP_USER[]   PROGMEM = "station@myftp.com";
const char FTP_PASS[]   PROGMEM = "mypass";

File imageFile;
int counter=0;

void setup() {
  // Iniciando Serial
  Serial.begin(115200);

  if(!FFat.begin(true)){
    pDBGln("Fat FS: Mount Failed");
    return;
  }else{
    pDBGln("Fat FS: Mount OK");
  }
  imageFile = FFat.open("/esp32pic.jpg"); 
  pDBGln("File mount OK.");  
}

void loop() {
  Serial.println("Loop starting....");  
  String myFile = "image" + String(counter) + ".jpg";
  uploadFile(myFile);
  pDBGln(myFile);
  delay(20000);
  counter++;
}

void uploadFile(String fileName) {
  String transitPacketBuffer = "";
  char singlePixelBuffer;
  int transitPacketBufferSize = 512;
  
  pDBG("Transfering file: ");
  pDBGln(fileName);  
  
  FTP ftp(GSM_BAUD_RATE, MODEM_RX, MODEM_TX, MODEM_RST);
  
  pDBGln("Loading FTP parameters...");
  
  ftp.putBegin(BEARER, fileName.c_str(), FTP_SERVER, FTP_USER, FTP_PASS);
  
  pDBGln("FTP connection ready"); 
  
  if (imageFile) {
    int i = 0;
    word bytesTransfered = 0;                                  
    word packetCounter   = 0;
    word fileSize        = imageFile.size();
    word numberOfPackets = fileSize/transitPacketBufferSize;
    if((fileSize%transitPacketBufferSize)>0){
      numberOfPackets++;
    }                                  
    while(imageFile.available()>0){
      singlePixelBuffer = imageFile.read();
      transitPacketBuffer.concat(singlePixelBuffer);
      i++;
      bytesTransfered++;
      if (i == transitPacketBufferSize) {
        ftp.putWrite(transitPacketBuffer.c_str(), transitPacketBufferSize);
        transitPacketBuffer = "";
        i = 0;      
        packetCounter++;
        delay(100); 
      }
    }
    if(transitPacketBuffer.length() != 0){
      ftp.putWrite(transitPacketBuffer.c_str(), i);
      packetCounter++;   
    }
    ftp.putEnd();
    imageFile.close();                                  
  }
}`

Console display:
`Fat FS: Mount OK
File mount OK.
Loop starting....
Transfering file: image0.jpg
Loading FTP parameters...
`

the image is preloaded.

As I said I have another sketch that uses AT commands and I can transfer this image (8K) to my FTP server but some 50% of the times it arrives incomplete (parts of it). 

I am using an ESP32 variant (TTGO Call that has a built in SIM800L).

Thanks in advance
Paulo

pcbtborges, 2020-05-13:
Hi, I've adapted my sketch to the link you provided as below:

```
#define DEBUG_ESP              //comment out to deactivate debug console

#ifdef DEBUG_ESP
  #define pDBGln(x) Serial.println(x)
  #define pDBG(x)   Serial.print(x)
#else 
  #define pDBG(...)
  #define pDBGln(...)
#endif

#include <Ftp.h>
#include "FFat.h"
#include "../examples/utils/logger.h"

#define GSM_BAUD_RATE 9600

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26

unsigned int RX_PIN = 7;
unsigned int TX_PIN = 8;
unsigned int RST_PIN = 12;
const char BEARER[]     PROGMEM = "claro.claro.com";
const char FTP_SERVER[] PROGMEM = "ftp.hidroflux.com";
const char FTP_USER[]   PROGMEM = "station@iot.hidroflux.com";
const char FTP_PASS[]   PROGMEM = "gen99ius";

File dataFile;
int counter=0;

void setup() {
  // Iniciando Serial
  Serial.begin(115200);

  if(!FFat.begin(true)){
    pDBGln("Fat FS: Mount Failed");
    return;
  }else{
    pDBGln("Fat FS: Mount OK");
  } 
}

void loop() {
  Serial.println("Loop starting....");  
  String myFile = "image" + String(counter) + ".jpg";
  Serial.println(myFile);
  uploadFile(myFile.c_str());
  Serial.println("transfer Sucessfull");
  delay(20000);
  counter++;
}

void uploadFile(const char *FTPimageFile) {
  dataFile = FFat.open("/esp32pic.jpg"); 
  Serial.println("File mount OK."); 

  if (dataFile) {
    FTP ftp(GSM_BAUD_RATE, RX_PIN, TX_PIN, RST_PIN);
    ftp.wakeUp();
    Serial.println("FTP wakeUp done.");
    ftp.putBegin(BEARER, FTPimageFile, FTP_SERVER, FTP_USER, FTP_PASS);       <<---FREEZES HERE
    Serial.println("ftp.putBegin Started.");
    unsigned int i;
    unsigned int chunkSize = 64;
    unsigned int writes = ceil(dataFile.size() / chunkSize);
    char buff[chunkSize];
   
    while (dataFile.available()) {
      i = 0;
      while (i < chunkSize) {
        buff[i] = dataFile.read();
        ++i;
      }

      if (ftp.putWrite(buff, i) != SUCCESS){
        return;
      }
      writes --;

      info(F("Pending: "), false);
      info(writes, false);
      info(F("/"), false);
      info(ceil(dataFile.size() / chunkSize), true);
    }

    ftp.putEnd();
    ftp.sleep();
    dataFile.close();
  }else {
    info(F("FTP. File open failed: "));
    info(FTPimageFile);
  }
}
```

It is still freezing as console printout below:

```
Fat FS: Mount OK
Loop starting....
image0.jpg
File mount OK.
FTP wakeUp done.
```

What do you think may be the problem?
Thanks again
Paulo

pcbtborges, 2020-05-13:
Hi, I've adapted my sketch to the link you provided as below:

```
#define DEBUG_ESP              //comment out to deactivate debug console

#ifdef DEBUG_ESP
  #define pDBGln(x) Serial.println(x)
  #define pDBG(x)   Serial.print(x)
#else 
  #define pDBG(...)
  #define pDBGln(...)
#endif

#include <Ftp.h>
#include "FFat.h"
#include "../examples/utils/logger.h"

#define GSM_BAUD_RATE 9600

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26

const char BEARER[]     PROGMEM = "claro.claro.com";
const char FTP_SERVER[] PROGMEM = "ftp.hidroflux.com";
const char FTP_USER[]   PROGMEM = "station@iot.hidroflux.com";
const char FTP_PASS[]   PROGMEM = "gen99ius";

File dataFile;
int counter=0;

void setup() {
  // Iniciando Serial
  Serial.begin(115200);

  if(!FFat.begin(true)){
    pDBGln("Fat FS: Mount Failed");
    return;
  }else{
    pDBGln("Fat FS: Mount OK");
  } 
}

void loop() {
  Serial.println("Loop starting....");  
  String myFile = "image" + String(counter) + ".jpg";
  uploadFile(myFile.c_str());
  Serial.println("transfer Sucessfull");
  delay(20000);
  counter++;
}

void uploadFile(const char *FTPimageFile) {
  dataFile = FFat.open("/esp32pic.jpg"); 
  
  Serial.println("File mount OK."); 
  Serial.print("FTPimageFile: "); 
  Serial.println(FTPimageFile); 
  
  if (dataFile) {
    FTP ftp(GSM_BAUD_RATE, MODEM_RX, MODEM_TX, MODEM_RST);
    ftp.wakeUp();
    Serial.println("FTP wakeUp done.");
    ftp.putBegin(BEARER, FTPimageFile, FTP_SERVER, FTP_USER, FTP_PASS);
    Serial.println("ftp.putBegin Started.");
    unsigned int i;
    unsigned int chunkSize = 64;
    unsigned int writes = ceil(dataFile.size() / chunkSize);
    char buff[chunkSize];
   
    while (dataFile.available()) {
      i = 0;
      while (i < chunkSize) {
        buff[i] = dataFile.read();
        ++i;
      }

      if (ftp.putWrite(buff, i) != SUCCESS){
        return;
      }
      writes --;

      info(F("Pending: "), false);
      info(writes, false);
      info(F("/"), false);
      info(ceil(dataFile.size() / chunkSize), true);
    }

    ftp.putEnd();
    ftp.sleep();
    dataFile.close();
  }else {
    info(F("FTP. File open failed: "));
    info(FTPimageFile);
  }
}
```

It is still freezing as console printout below:

```
Fat FS: Mount OK
Loop starting....
File mount OK.
FTPimageFile: image0.jpg
FTP wakeUp done.
```

What do you think may be the problem?
Thanks again
Paulo

carrascoacd, 2020-05-14:
Hi Paulo, could you uncomment [this line](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Sim800.h#L40) and share the serial output again, please?

I think your SIM800L module is not powered, have you tried this?

```
  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);
```

I think your board can enable/disable SIM800L in that way.

pcbtborges, 2020-05-14:
Hi Antonio, thanks again for your help.

In fact I forgot to include the lines you pointed out.
I did include them as indicated but now the system reboots continuously as console printout below:

```
Fat FS: Mount OK
Loop starting....
File mount OK.
Modem init OK
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40081160  PS      : 0x00060031  A0      : 0x800812c0  A1      : 0x3ffbe770  
A2      : 0x08000000  A3      : 0x00000000  A4      : 0x0000001b  A5      : 0x00000000  
A6      : 0x3fefffff  A7      : 0xffffff80  A8      : 0x3ffb8a40  A9      : 0x3ffbe750  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc04ec  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000005  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40081160  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40088f4f

Backtrace: 0x40081160:0x3ffbe770 0x400812bd:0x3ffbe790 0x4008138d:0x3ffbe7b0 0x40086ebd:0x3ffbe7d0 0x400f4e23:0x3ffbc5e0 0x400d63d7:0x3ffbc600 0x4008bcf2:0x3ffbc620 0x4008a801:0x3ffbc640

Rebooting...
```

The Adjusted sketch is as below:

```
#define DEBUG_ESP              //comment out to deactivate debug console

#ifdef DEBUG_ESP
  #define pDBGln(x) Serial.println(x)
  #define pDBG(x)   Serial.print(x)
#else 
  #define pDBG(...)
  #define pDBGln(...)
#endif

#include <Ftp.h>
#include "FFat.h"
#include "../examples/utils/logger.h"

#define GSM_BAUD_RATE 9600

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26

const char BEARER[]     PROGMEM = "claro.claro.com";
const char FTP_SERVER[] PROGMEM = "ftp.myftpserver.com";
const char FTP_USER[]   PROGMEM = "station@myftpserver.com";
const char FTP_PASS[]   PROGMEM = "mypass";

File dataFile;
int counter=1;

void setup() {
  // Init Serial
  Serial.begin(115200);

  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);  

  if(!FFat.begin(true)){
    pDBGln("Fat FS: Mount Failed");
    return;
  }else{
    pDBGln("Fat FS: Mount OK");
  } 
}

void loop() {
  pDBGln("Loop starting....");  
  String myFile = "image" + String(counter) + ".jpg";
  char charFileName[myFile.length()+1];
  myFile.toCharArray(charFileName,myFile.length()+1);
  uploadFile(charFileName);
  pDBGln("transfer Sucessfull");
  delay(20000);
  counter++;
}

void uploadFile(const char *FTPimageFile) {
  dataFile = FFat.open("/esp32pic.jpg");  
  pDBGln("File mount OK.");  
  if (dataFile) {
    FTP ftp(GSM_BAUD_RATE, MODEM_RX, MODEM_TX, MODEM_RST);
    pDBGln("Modem init OK");
    ftp.wakeUp();
    pDBGln("FTP wakeUp done.");
    pDBG("Configuring FTP. \nBearer: ");
    pDBG(BEARER);
    pDBG(", Image: ");
    pDBG(FTPimageFile);
    pDBG(", size: ");
    pDBG(dataFile.size());
    pDBG(",\nsrv: ");
    pDBG(FTP_SERVER);
    pDBG(",\nuser: ");
    pDBG(FTP_USER);
    pDBG(", pass: ");
    pDBGln(FTP_PASS);
    
    pDBGln("Starting ftp.putBegin ...");
    ftp.putBegin(BEARER, FTPimageFile, FTP_SERVER, FTP_USER, FTP_PASS);
    pDBGln("ftp.putBegin OK.");
    
    unsigned int i;
    unsigned int chunkSize = 512;
    unsigned int writes = ceil(dataFile.size() / chunkSize);
    char buff[chunkSize];
   
    while (dataFile.available()) {
      i = 0;
      while (i < chunkSize) {
        buff[i] = dataFile.read();
        ++i;
      }

      if (ftp.putWrite(buff, i) != SUCCESS){
        return;
      }
      writes --;

      info(F("Pending: "), false);
      info(writes, false);
      info(F("/"), false);
      info(ceil(dataFile.size() / chunkSize), true);
    }

    ftp.putEnd();
    ftp.sleep();
    dataFile.close();
  }else {
    info(F("FTP. File open failed: "));
    info(FTPimageFile);
  }
}
```

Thanks again
Paulo

pcbtborges, 2020-05-14:
Hi Antonio, thanks again for your help.

In fact I forgot to include the lines you pointed out.
I did include them as indicated but now the system reboots continuously as console printout below:

```
Fat FS: Mount OK
Loop starting....
File mount OK.
Modem init OK
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40081160  PS      : 0x00060031  A0      : 0x800812c0  A1      : 0x3ffbe770  
A2      : 0x08000000  A3      : 0x00000000  A4      : 0x0000001b  A5      : 0x00000000  
A6      : 0x3fefffff  A7      : 0xffffff80  A8      : 0x3ffb8a40  A9      : 0x3ffbe750  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc04ec  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000005  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40081160  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40088f49

Backtrace: 0x40081160:0x3ffbe770 0x400812bd:0x3ffbe790 0x4008138d:0x3ffbe7b0 0x40086ebd:0x3ffbe7d0 0x400f4e2f:0x3ffbc5e0 0x400d63e3:0x3ffbc600 0x4008bcf2:0x3ffbc620 0x4008a801:0x3ffbc640

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Fat FS: Mount OK
Loop starting....
File mount OK.
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400810b4  PS      : 0x00060034  A0      : 0x800812c0  A1      : 0x3ffbe770  
A2      : 0x3ffb1ef4  A3      : 0x554ed296  A4      : 0x00000000  A5      : 0x00000008  
A6      : 0x15543884  A7      : 0xffffff80  A8      : 0x044b89c8  A9      : 0x00000000  
A10     : 0x3ffb9a18  A11     : 0x3ffb8a9c  A12     : 0x8008d142  A13     : 0x3ffb1d20  
A14     : 0x00000001  A15     : 0x3ffb8a9c  SAR     : 0x00000013  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  
Core 1 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400810b4

Backtrace: 0x400810b4:0x3ffbe770 0x400812bd:0x3ffbe790 0x4008138d:0x3ffbe7b0 0x40086ebd:0x3ffbe7d0 0x4000bfed:0x3ffb1de0 0x4008b681:0x3ffb1df0 0x4008236e:0x3ffb1e10 0x400d2fc1:0x3ffb1e30 0x400d2fd9:0x3ffb1e60 0x400d1d4d:0x3ffb1e80 0x400d1de1:0x3ffb1ea0 0x400d13b9:0x3ffb1ec0 0x400d1672:0x3ffb1f50 0x400d3bb5:0x3ffb1fb0 0x4008a801:0x3ffb1fd0

Core 0 register dump:
PC      : 0x400f4e32  PS      : 0x00060134  A0      : 0x800d63e6  A1      : 0x3ffbc060  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800d5fae  A9      : 0x3ffbc030  
A10     : 0x00000000  A11     : 0x40087648  A12     : 0x00060120  A13     : 0x3ffbb9e0  
A14     : 0x00000000  A15     : 0x3ffbbd60  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400f4e32:0x3ffbc060 0x400d63e3:0x3ffbc080 0x4008bcf2:0x3ffbc0a0 0x4008a801:0x3ffbc0c0

Rebooting...

```

The Adjusted sketch with the is as below:

```
[... 97 lines cut ...]
      }

      if (ftp.putWrite(buff, i) != SUCCESS){
        return;
      }
      writes --;

      info(F("Pending: "), false);
      info(writes, false);
      info(F("/"), false);
      info(ceil(dataFile.size() / chunkSize), true);
    }

    ftp.putEnd();
    ftp.sleep();
    dataFile.close();
  }else {
    info(F("FTP. File open failed: "));
    info(FTPimageFile);
  }
}

```
Thanks again
Paulo

carrascoacd, 2020-05-15:
I can´t see the debug traces, think I have to add a debug trace for ESP32, [this is not working](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Sim800.cpp#L126), so change that line by `pDBG`. I mean:

```
 Serial.print(c);
```
by
```
pDBG(c)
```

pcbtborges, 2020-05-15:
Hi, I did the change (pDBG(c)) but the sketch fails to compile. Error compiling for ESP32.

I believe it might be related to the way serial is assigned.
The test sketch below works. Please see the way the Serial is assigned.
Could that be the reason it fails to init the serial?
I will keep trying to understand the logic and inform if I find something relevant.
Thanks

```
#define SerialAT Serial1                          <<<-------------------Assign serial1

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26

void setup() {
  Serial.begin(115200);
  // Set modem reset, enable, power pins
  pinMode(MODEM_PWKEY, OUTPUT);
  pinMode(MODEM_RST, OUTPUT);
  pinMode(MODEM_POWER_ON, OUTPUT);
  digitalWrite(MODEM_PWKEY, LOW);
  digitalWrite(MODEM_RST, HIGH);
  digitalWrite(MODEM_POWER_ON, HIGH);

  // Set GSM module baud rate and UART pins
  SerialAT.begin(9600, SERIAL_8N1, MODEM_RX, MODEM_TX);
  delay(10000);
}

void loop() {
  Serial.println("Start modem...");
  sendCommand("at+cfun?");
  sendCommand("AT+SAPBR=3,1,\"Contype\",\"GPRS\"");
  sendCommand("AT+SAPBR=3,1,\"apn\",\"claro.claro.com\"");
  sendCommand("at+sapbr=1,1");
  
  //FTP
  sendCommand("AT+FTPCID=1");
  sendCommand("AT+FTPSERV=\"myftpserver.com\"");
  sendCommand("AT+FTPPORT=21");
  sendCommand("AT+FTPUN=\"station@myftpserver.com\"");
  sendCommand("AT+FTPPW=\"mypass\"");
  sendCommand("AT+FTPPUTNAME=\"mytestimage.jpg\"");
  sendCommand("AT+FTPPUTPATH=\"/\"");
  sendCommand("AT+FTPSIZE");

  if(chkFileSize()){
    Serial.println("OK");
  }else{
    Serial.println("NOT OK");
  }
  Serial.println("END...");
  delay(15000);
}

bool chkFileSize(){
  if(sendCommand("at+cfun?")==0){return false;}
  if(sendCommand("AT+SAPBR=3,1,\"Contype\",\"GPRS\"")==0){return false;}
  if(sendCommand("AT+SAPBR=3,1,\"apn\",\"claro.claro.com\"")==0){return false;}
  if(sendCommand("at+sapbr=1,1")==0){return false;}  
  if(sendCommand("at+httpinit")==0){return false;}
  if(sendCommand("at+httppara=\"cid\",1")==0){return false;}
  if(sendCommand("at+httppara=\"url\",\"http://myserver.com/upload/chkfsize.php?filename=acid.jpg\"")==0){return false;}
  if(sendCommand("at+httpaction=0")==0){return false;}
  if(sendCommand("at+httpread")==0){return false;} 
  return true; 
}

byte sendCommand(String serialCommand){
  String content = "";
  char character;  
  byte chkStatus=99;
  SerialAT.println(serialCommand);
  Serial.print(serialCommand);
  Serial.println(": ");
  delay(200);
  while(SerialAT.available()>0) {
    character = SerialAT.read();
    content.concat(character);
    delay(30);
    //Stop reading conditions
    if (content.indexOf("ERROR") != -1){
      chkStatus = 0;
      break;
    }
  }
  Serial.print(content);
  delay(3000);
  return chkStatus;
}
```

pcbtborges, 2020-05-15:
Hi, some more tests.
Changes made on SIM800.ccp:
```
int SIM800::preInit(void)
{                                  
    pinMode(resetPin, OUTPUT);
					Serial.println("resetPin set to OUTPUT");
    digitalWrite(resetPin, HIGH);
    delay(200);
					Serial.println("resetPin set HIGH");
    digitalWrite(resetPin, LOW);
    delay(2000);
					Serial.println("resetPin set to LOW");
    digitalWrite(resetPin, HIGH);
    delay(3000);
					Serial.println("resetPin set to HIGH");
    purgeSerial();
					Serial.println("purgeSerial OK");
    serialSIM800.flush();
					Serial.println("flush OK");
    return TRUE;
}
```

Console output:

```
Fat FS: Mount OK
Loop starting....
File mount OK.
Modem init OK
resetPin set to OUTPUT
resetPin set HIGH
resetPin set to LOW
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
```

carrascoacd, 2020-05-15:
My library uses the `SofwareSerial` to communicate with SIM800L, but you are using `HardwareSerial` with this variable `Serial1`.

Try changing [this line](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Sim800.h#L142)

```
SoftwareSerial serialSIM800;
```
by
```
HardwareSerial serialSIM800;
```

Then I think you will need to change the constructor as well, so modify [this line](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Sim800.h#L58):

```
unsigned int rstPin) : serialSIM800(txPin, rxPin)
```
by
```
unsigned int rstPin) : serialSIM800(1)
```

then [this line](https://github.com/carrascoacd/ArduinoSIM800L/blob/master/src/Sim800.h#L60)
```
serialSIM800.begin(baudRate);
```
by
```
serialSIM800.begin(baudRate, SERIAL_8N1, txPin, rxPin);
```

carrascoacd, 2020-05-15:
Could you share your results in this new issue https://github.com/carrascoacd/ArduinoSIM800L/issues/43 I think we are talking about a different topic here now

closed 2020-05-15 by carrascoacd

UssamaZafar, 2024-12-04:
I want todo something similar to upload an image from ARDU cam and upload in on a FTP server using Sim800L + ESP32.

Can anyone help me with this. I am really bad with programming.