carrascoacd/ArduinoSIM800L #46: Can it send large .txt files of like 10 MB?

bhupiister, 2020-08-23:
Can it send large .txt files of like 10 MB? 
I will try this code on ESP32 board. Since its working on Arduino, it should work on esp32 too.
I want to store sensor data all day and at night want to push this sensor data on server via FTP of HTTP post. Which is the recommended method ?

carrascoacd, 2020-08-23:
I'd say that FTP is better in SIM800L: easier.

Try with small files instead, divide the 10mb file in small files and send
them more often.

bhupiister, 2020-08-23:
I wanted some library that could do this division part. 
After putting this command 
AT+FTPPUT=1
OK
+FTPPUT: 1,1,1360
Does this mean that maximum bytes that can be sent is 1360?
How can we divide a file then?

bhupiister, 2020-08-23:
Came across this code today for large files, will test it. See if you can get something from it to improve your library.
`#include <SD.h>
#define csSD 5

bool debug = 1;
bool roaming = false;
char* file_name = /*"Filename"*/;
File dataFile;
File dataFileCopy;
const int Sim900PowerPin = 46;
String apn_name = /*"apn_name"*/;
String apn_username = ""; //not tested
String apn_password = ""; //not tested
String ftp = /*"ftp adress*/";
String ftp_user = /*"ftp username"*/;
String ftp_password = /*"ftp password*/";

void setup() {
  if (debug) {Serial.begin(19200);}
  Serial3.begin(19200);
  pinMode(csSD, OUTPUT);

  //init SD Card
  byte answer = 0;
  int i = 0;
  while (i < 5 && answer == 0) {
    if (!SD.begin(csSD)) {
      if (debug) {Serial.println("SD Card initialization failed!");}
      i++;
      delay(500);
    }
    else {
      if (debug) {Serial.println("SD Card initialization done.");}
      answer = 1;
    }
  }
  if (answer == 0) {/*Send SMS that init SD Card is failed*/}
  //done init SD card

  //check if file arivle
  else if(answer == 1) {answer = checkFile();}
  if (answer == 0) {/*Send SMS that file is not avriable*/}

  //GPRS part
  else if(answer == 1) {answer = powerUp_Sim900();}
  if (answer == 0) {/*Send SMS that powerUP is failed*/}
  else if (answer == 1) {delay(20000); answer = initSim900();} //Delay to let startuo the SIM900 Module
  if (debug) {Serial.println("--- Answer initSim900: " + String(answer));}
  if (answer == 0) {/*Send SMS that Sim900 init failed*/}
  else if (answer == 1) {answer = initSim900FTP();}
  if (debug) {Serial.println("--- Answer initSim900FTP: " + String(answer));}
  if (answer == 0) {/*Send SMS that Sim900FTP init failed*/}
  else if (answer == 1) {answer = Sim900FTPSend();}
  if (debug) {Serial.println("--- Answer Sim900FTPSend: " + String(answer));}
  if (answer == 0) {/*Send SMS that Sim900FTPSend failed*/}
}

void loop() {
 
}

byte initSim900() {
  byte answer = 0;
  int i = 0;
  while (i < 10 && answer == 0){
    answer = sendATcommand("AT+CREG?","+CREG: 0,1", 1000);
    i++;
    delay(1000);
  }
  if (roaming and answer == 0) {
[... 180 lines cut ...]
  return answer; //send 0 if power up
}

void powerSim900() {
  pinMode(Sim900PowerPin, OUTPUT);
  digitalWrite(Sim900PowerPin,LOW);
  delay(1000);
  digitalWrite(Sim900PowerPin,HIGH);
  delay(2000);
  digitalWrite(Sim900PowerPin,LOW);
  delay(5000);
}

char* string2char(String command){
  if(command.length() != 0){
    char *p = const_cast<char*>(command.c_str());
    return p;
  }
}

byte checkFile() {
  byte answer = 0;
  if(SD.exists(file_name)) {answer = 1;}
  return answer;
}`

carrascoacd, 2020-08-23:
It means that the max chunk size is 1360 bytes per request.

Try writing small files (some Kb) more often and upload them. Do not try to
divide the 10mb one.

basurilla82, 2022-11-27:
Hello, have you solve this problem?. I want to send a big picture file by ftp with sim800l and esp32 (not wifi). Please, can you write your final code source?. Thanks.