carrascoacd/ArduinoSIM800L #48: Aplications for STM32 using HardwareSerial Serial1, Serial2 and Serial3

viniciusbarozzi, 2020-10-25:
Hello, I'm working on the library changes to be able to work with the STM32 on Arduino Ide, but I can't find a way to identify which serial port will be used. I'm using as a base the version modified for ESP32 where some changes have already been made to work with HardwareSerial, but I can't understand where to define which Serial port to use.

I appreciate any contribution.
Thank you.

carrascoacd, 2020-11-01:
Hello, you can use SoftwareSerial and it should work properly.

viniciusbarozzi, 2020-11-01:
Hello Antônio, I tried to use SoftwareSerial with Stm32, but it doesn't work because in the stm32duino repository there is no SoftwareSerial Function precisely because it already has 3 hardware serial ports.  I tried to modify the construction method of your library, but I couldn't, finally, started to develop a program based on the procedures used in your library.  My focus is only to use the POST function over HTTP.  If you are interested, I can share here the development of the program I am doing.

carrascoacd, 2020-11-02:
Ok, I understand. 

Feel free to see this branch https://github.com/carrascoacd/ArduinoSIM800L/tree/add-compatibility-for-esp32 where I did some modifications to support it. Take a look especially at this file and line https://github.com/carrascoacd/ArduinoSIM800L/blob/add-compatibility-for-esp32/src/Sim800.h#L60 

You will have to initialize your serial variable like this I think https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/HardwareSerial.cpp#L37

If you manage to make it work we can add some configuration to support both software and hardware serial.